sbwml/builder #1: r5s最新固件kms服务无法从web页面上启动

duguying, 2022-11-01:
固件版本：**OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a**
硬件：**r5s**

界面上kms勾选启动，点击应用，无法启动服务。但是在终端中可以通过命令启动，自己添加了 `/etc/init.d/vlmcsd` 也无法通过web界面启动

sbwml, 2022-11-01:
### 测试正常，自行检查端口是否被其它程序占用

[image]
[image]

duguying, 2022-11-01:
端口没有被占用，因为我从终端能启动，终端启动后，发现web上重新启动，会导致关闭成功但是启动失败。

sbwml, 2022-11-01:
这个我不清楚了     在我测试中并没有复现这个问题

duguying, 2022-11-01:
web操作有地方看日志吗？点击应用按钮后触发的是什么命令

duguying, 2022-11-01:
我看你的版本跟我的有点区别。不知道是否我这个版本有这个问题

sbwml, 2022-11-01:
我不知道，因为我是一个小趴菜。

但是你可以尝试这些：

1、这个命令如果有任何反馈，并且名称不是  vlmcsd，那就是端口冲突
```shell
lsof -i:1688
```

2、这个命令如果有报错，我建议贴信息上来
```shell
/etc/init.d/kms restart
```

3、如果第二个命令没有报错，执行这个命令有输出说明插件没问题，如果没有输出参考 **4、**
```shell
ps | grep vlmcsd | grep -v grep
```

4、如果没有任何输出，就继续执行下一个命令行。如果这个命令它出错了，说明op系统内的配置被乱改过了，我也不知道怎么弄，因为我是小趴菜

```shell
/usr/bin/vlmcsd -i /etc/vlmcsd/vlmcsd.ini -L 0.0.0.0:1688
```

继上如果没有报错，执行这个，若它有任何输出，说明插件没问题

```shell
ps | grep vlmcsd | grep -v grep
```

sbwml, 2022-11-01:
另外ps：有种科学而又不科学的测试方法，启用KMS 插件。重启系统

duguying, 2022-11-01:
的确，重启路由器后，能启动成功

duguying, 2022-11-01:
多谢大佬

closed 2022-11-01 by duguying