sbwml/builder #7: 24.10固件bug多到离谱

LeaoLi8, 2025-03-11:
passwall设置负载均衡后无法启动，ddns无法自启运行

sbwml, 2025-03-11:
# 你除了可以去换一个好用的固件外，我没有任何一点建议能给到你。

[image]

[image]

[image]

LeaoLi8, 2025-03-12:
你截的是r5s么

sbwml, 2025-03-13:
# 是 R5S，但这里的固件内置插件都是一样的源码，即使 x86_64 也一样。

> passwall设置负载均衡后无法启动，ddns无法自启运行

# 所以你能做的，只有换一个固件。比如稳定的 LEDE。

LeaoLi8, 2025-03-15:
@wx2020 你是版主养的狗么

xianren78, 2025-03-20:
[image]

这儿没有版主，有没有乱咬人的**狗**就不好说了，sbw大佬在 https://github.com/sbwml/r4s_build_script 首页挂过一条，有兴趣可以移步参观。