sbwml/builder #11: 大佬你好可以请教一下怎么官方源码24.10编译的时候把6.12的内核融合进去？

wixxm, 2025-03-30:
大佬如题，打扰了。

sbwml, 2025-04-02:
1、6.12 通用补丁
2、6.12 设备补丁
3、include/kernel-6.12 内核版本信息
4、手动修复一切与 6.12 内核构建失败的包

你知道我在说什么吗？反正我是不知道。

这些东西就根本不是文字能描述出来的。建议去 lede 仓库看看 6.12 内核是如何被加入的，这肯定受益良多。

wixxm, 2025-04-02:
我把lede的关于6.12的文件都复制到openwrt官方的24.10源码里边了，然后编译失败。。

sbwml, 2025-04-02:
很正常，openwrt 和 lede 本身就差别巨大。这不是复制粘贴就能完成的操作。你应该去搜搜有没有人fork 了 openwrt 源码并且加入了 6.12 内核、

wixxm, 2025-04-02:
> 如果你想要的是 24.10 固件 + 6.12 内核。那么这里有现成的：https://github.com/sbwml/builder/releases
> 
> 如果你想要的是可编译的源码，我则没有。毕竟没有一点精力去维护一个 openwrt 分支。

感觉就是很复杂 要改的东西很多，单纯该内核配置估计不行 确实何难维护

sbwml, 2025-04-02:
你早俩个月前来问，倒是可以给到你一个标准的流程来抄。不过现在没有了，因为社会险恶不再开源

https://github.com/sbwml/r4s_build_script?tab=readme-ov-file#%E6%8E%88%E6%9D%83%E6%9E%84%E5%BB%BA

wixxm, 2025-04-07:
> 你早俩个月前来问，倒是可以给到你一个标准的流程来抄。不过现在没有了，因为社会险恶不再开源
> 
> https://github.com/sbwml/r4s_build_script?tab=readme-ov-file#%E6%8E%88%E6%9D%83%E6%9E%84%E5%BB%BA
终有一些人有一些奇怪的想法，感谢大佬指点，请大佬喝奶茶，我关闭啦。

closed 2025-04-07 by wixxm

qingtian110, 2025-04-08:
友情赞助五十大洋你值得拥有  

https://github.com/qingtian110/r4s_build_script/actions/runs/14333494597/job/40174720201

wixxm, 2025-04-08:
> 友情赞助五十大洋你值得拥有
> 
> https://github.com/qingtian110/r4s_build_script/actions/runs/14333494597/job/40174720201

那就是我赞助的不够 我再去添添