sbwml/builder #37: write image to SD card

r0mata-gh, 2025-09-22:
Hello, 

I have a question regarding your firmware build. 

I recently received a Nanopi R76S.
As the first step, I erased the eMMC since I don't plan to use it.
FriendlyWRT comes with an older kernel, and that's when I found your custom builds. 

I downloaded the images, verified their checksums, and tried flashing them to an SD card using balenaEtcher 2.1.4. However, the board fails to boot from the SD card — while the original FriendlyWRT image flashes and boots successfully on the same card. 

Another concern: when testing or extracting the archive with 7-Zip, it shows a warning:
"There are some data after the end of the payload data: openwrt-24.10.3-rockchip-armv8-friendlyarm_nanopi-r76s-squashfs-sysupgrade.img"
Also, the extracted file size does not match the size shown when opening the archive. 

On Windows, MultiCommander extracts the image without warnings, and the resulting size appears correct — but still, the device won't boot.
On Linux, GUI archive managers (like Archive Manager) report errors when opening or extracting the files. 

I've tried flashing both the .gz file directly and the extracted .img file using Etcher — result is the same.
I've also tested different versions — same issue. 

Affected images: 

    openwrt-24.10.2-rockchip-armv8-friendlyarm_nanopi-r76s-squashfs-sysupgrade.img.gz
    openwrt-24.10.3-rockchip-armv8-friendlyarm_nanopi-r76s-squashfs-sysupgrade.img.gz
     

Could you please advise what might be going wrong? Am I doing something incorrectly?

sbwml, 2025-09-22:
Because I’m using mainline U-Boot to generate the idbloader, and it’s a known issue that mainline U-Boot currently cannot boot the RK3576 from an SD card. To solve this problem, the idbloader needs to be generated by **boot_merger**, or use **Rockchip’s U-Boot**. For standard RK3576 development boards, this is usually straightforward.

However, the situation I’m facing is that even when I use the original FriendlyWRT image, the NanoPi R76S still fails to boot from an SD card. This is what’s really preventing me from moving forward.

Right now I’m using the NanoPi R76S with 16GB (LPDDR5) RAM and 32GB eMMC. I’m not sure whether the LPDDR5 memory makes a difference here.

> while the original FriendlyWRT image flashes and boots successfully on the same card.

What specifications does your R76S have?

sbwml, 2025-09-22:
Also, at the moment my firmware definitely cannot boot solely from the SD card. It only works if it is written to the eMMC, or if the eMMC already contains a valid bootloader—only then can it boot from the SD card.

Until I find a method or firmware that allows my own device to boot from the SD card with the eMMC erased, this firmware will remain as it is. Alternatively, you might want to try looking for firmware from other developers.

r0mata-gh, 2025-09-24:
Of course, but my knowledge is quite limited. However, I'm ready to help as much as I can.

sbwml, 2025-09-25:
> Of course, but my knowledge is quite limited. However, I'm ready to help as much as I can.

https://r76s.cooluc.com/d/minimal/openwrt-24.10/v24.10.3/openwrt-24.10.3-rockchip-armv8-friendlyarm_nanopi-r76s-squashfs-sysupgrade.img.gz

Can you try this firmware?

closed 2025-10-02 by r0mata-gh